probe-rs/rusty-probe #1: Form factor consideration

code-db, 2022-04-09:
For the rusty-probe to be used as a mezzanine board on a base card, it would ideally be soldered to a PCB adapter with a standard pitch. Has an adapter PCB been made? What considerations would that adapter/rusty-probe require?

The bottom of the rusty-probe PCB contains configuration details in the silk screen and exposed copper pads. That information & functionality would no longer be accessible if the rusty-probe was used on an adapter PCB.

korken89, 2022-11-28:
The solder-in option is if you want to have it on e.g. a production unit, this is how we use the solder in option.
Plus the STLinkv3 does similarly (where we got the idea from).

However no more thought than that has been put into this.

korken89, 2022-11-29:
Oh yeah, another point on this.
If one does not want to pay for the castellated vias, it's fine to just delete them before sending to production as well.

closed 2024-11-10 by korken89