probe-rs/rusty-probe #4: JLC assembly compatible BOM

allexoll, 2022-09-12:
having a BOM compatible with JLC assembly service would be pretty helpfull for accessibility.

Might have a look later if I find the time.

korken89, 2022-11-28:
It would be awesome! I'd also like there to be such, ping me if you start looking and I can give feedback.
I think even if we need to update the design it would be worth it.

allexoll, 2022-11-28:
#6 should be a start, but I have not ordered some using that ( no stock on any semiconductor anyway...)

korken89, 2022-11-28:
Indeed, I have no idea how lcsc works so finding a suitable LDO should be the only thing that needs fixing right?

allexoll, 2022-11-28:
no, what I meant is that when I ordered my small batch, they did not have any ic in stock, mostly due to shortage. From what i understand from JLC/LCSC, you can order the parts in advance for them to have them in stock when you assemble it. it just was not worth it for the 10 I built, and for those I only asked them to place the passive (Rs, Cs) and a few others: Osc, type-c connector, and so on. see photo: 
[image]

I have assembled one with components I ordered from mouser, but I did not have time to test it yet.

korken89, 2022-11-28:
Ahhh, I understand. Looks great!

FYI - I just pushed updates to the design so we can use the active discharge version of NCP133 (which seems to be easy to get).
It should not impact the code though, so this board should work fine in the end.

korken89, 2022-11-28:
Also I added CI actions to generate gerbers, images of PCB etc for future use.

korken89, 2024-11-10:
I gave this a look again, there seems to be a lot better part selection on JLCPCB now - I'm quite sure one can adapt the board into an JLCPCB friendly version.

berkus, 2024-12-06:
We have a JLCPCB bom now, so I guess this can be closed?