probe-rs/rusty-probe #7: Readme link to shop is broken

tgross35, 2023-01-21:
Hello,

It looks like the link in the readme, https://shop.probe.rs/, is broken

It comes up with a security warning then 403 forbidden if you continue to navigate to the site

Yatekii, 2023-01-21:
I disabled the shopify shop to save money while we cannot produce more probes due to chip shortage. I updated the README, thank you :)

tgross35, 2023-01-21:
Oh no, well thanks for the context.

Out of curiosity, what are the blocker parts? Looks like the RP2040 is back in stock on Digi https://www.digikey.com/en/products/detail/raspberry-pi/SC0908-7/14306009?src=raspberrypi, I don't think anything else would be beyond an easy alt

korken89, 2023-01-21:
The link is for the previous version based on STM32, this probe is still under testing. :)

As soon as it's ready we'll add the link back!

closed 2023-01-21 by korken89

tgross35, 2023-01-21:
Oh cool, I didn’t realize there was an updated part!

Fwiw if you ever need an extra hand for a schematic review / sourcing / whatever, feel free to ping me since that’s my day job (know it’s unlikely for this tiny project, but 🤷 doesn’t hurt)

Yatekii, 2023-01-21:
Reviews are always welcome and new hardware projects & improvements :) I think the hardware team atm is pretty much just @korken89 even tho we have some others contributing :)

The entire project lives of OS contributions, so please feel free to contribute whenever you feel like it :)

korken89, 2023-01-22:
@tgross35 hit me up on matrix with name and shipping address and I'll send you a prototype. :)

commit 62b25ed3 (2023-01-21): Fix #7