probe-rs/rusty-probe #12: Should the DTR/RTS pins of the virtual serial port be adapted

HalfSweet, 2024-01-01:
For some MCUs that rely on using the serial port for downloads, the use of RTS/DTR for auto-reset into their own bootloader is clearly needed, and for the debugger, we just need to add two more IO ports for output. Will we consider doing this in the rusty-probe later on?

Yatekii, 2024-01-01:
I am not sure what exactly you want, but if you want to flash via serial, you can use a serial adapter? :)

HalfSweet, 2024-01-01:
Yes, but I think rusty-probe could be more versatile if it had full USB to UART functionality.

For some MCUs that need UART for flash, such as the ESP32 the general practice is to use the RTS and DTR signals from the serial port as trigger flags

korken89, 2024-11-10:
That's a good future upgrade path, I'll keep it in mind!