probe-rs/rusty-probe #13: shop info, more shipping options

WizardUli, 2024-01-24:
It would be nice for the shop to have some more info like e.g. where it is shipping from. I suppose it is not Europe and in that case it would be nice to have shipping options like DHL Express which could handle taxes and customs duty for me.

Yatekii, 2024-02-01:
You can let me know via email or on Matrix if you want special shipping. But it's more effort for me to ship with a different provider and I am not keen for more work ^^

WizardUli, 2024-02-01:
I've already ordered two probes accepting I'll have to deal with VAT and duties and horrible Czech Post which usually takes more time to deliver something inside CZ than it takes for it to arrive into CZ even from the other side of the globe.

closed 2024-02-01 by WizardUli

Yatekii, 2024-02-01:
2 probes fly under the VAT limit in Switzerland (where I ship from). Nut sure what the limit is for CZ.