probe-rs/rusty-probe #15: Target support of "any" is too ambiguous and incorrect. List of supported debugging protocols/interfaces would be better.

acedogblast, 2024-04-30:
The comparison chart on the README has "any" for what targets are supported for all probes listed other than the STLinkV3. This is too broad and also not true. A list of supported debugging protocols/interfaces will provide a more accurate description of the capabilities of the probes.

From what I understand the Rusty Probe currently only supports JTAG and SWD (Serial Wire Debug for ARM chips) with support for Serial Wire Output (SWO) and Virtual COM Port (VCP) a WIP. Also where is UART support?

These protocols/interfaces are the most common but lesser known protocols like WCH's single wire debug (for RISC-V CH32Vxxx micro-controllers) is not included. Therefor the target support value of "any" is incorrect and misleading.

I can send a pull request to correct this error.

acedogblast, 2024-04-30:
What probe is "PicoProbe"? It it just a Raspberry PI Foundation's Pico line of RP2040 dev boards flashed with the Raspberry Pi Debug Probe firmware?

korken89, 2024-11-10:
Sorry, this issues and PR got forgotten. Can you update the PR from conflicts and I'll merge it?

acedogblast, 2024-11-11:
It seems there has not been any changes to master since I submitted the pull. So I just kept my changes.

closed 2024-11-14 by korken89