probe-rs/rusty-probe #17: Understanding the Target voltage feedback circuit and possible modification for 5V targets.

acedogblast, 2024-05-01:
Hello I wanted to ask about the functionality of the target voltage feedback circuit. I consider my knowledge in electrical engineering is mostly in the intermediate hobbyist level. From what I can understand in the schematic and reading the data sheet of the LDOs used, the feedback circuit controls the output voltage of the LDO. Traditionally it is a simple voltage divider circuit with simple algebra math to calculate the resistor values. According to the LDO's datasheet the feedback voltage is 0.8V. With just the top and bottom resistors (47K and 17.4K) the output voltage would be 2.96V. 
However in this design we have an additional voltage source provided by the PWM signal from the RP2040. This PWM signal is passed through 2 low-pass RC filters in series to convert PWM to a DC signal with a range between 0.0V to 3.3V forming a rudimentary digital to analogue converter. This is where my EE knowledge is insufficient on how this DC signal is handled. 

My best guess is this DC signal is like another "top voltage feed" for another voltage divider that shares the bottom resistor. Lets use R14, R15, and R16 in the Rusty Probe for analysis. The label "T_VCC" is both the output of U14 LDO and possibly the main power from the target. The label "VS_T" is the DC signal from the PWM DAC. Lastly the label "ADJ_T" is the midpoint of the voltage divider and is connected to the feedback pin of the LDO.

Initially the LDO would provide 2.96V which is lower than 3.3V (assuming that is the target's voltage). The RP2040 senses the lower voltage though a separate voltage divider and starts to create a lower DC output from the DAC circuit (lets say 2.6V). I am assuming that this 2.6V signal is able to overpower the original 2.96V because it sinks the current which results in a lower than 0.8V feedback signal to the LDO. This results in the LDO to increase the output voltage to compensate.

Now to the calculations of the voltage divider resistors. I do not have matlab nor have any experience using it or reading matlab code. Therefor I am asking for your help on understanding the calculations needed to pick the right resistor values.

The original design of the Rusty Probe is for targets with a signal voltage at or below 3.3V it would nice to make a version that can also support 5V signals. I have read the data sheets of the LDOs, the level shifters, and the ESD/EMI part U4. It would seem that the only limitation with the current BOM is that the LDOs have a maximum output of 3.6V. A possible replacement LDO that I have found is Diodes Inc. AP7366EA-SN-7 as it has very similar characteristics. It has the same 0.8V feedback voltage as the existing LDOs (NCP133AMXADJTCG) and a lower Iq current of 60uA. Dropout voltage is a bit higher at the same output currents but should be fine since 5V high signals have a larger tolerance range.

I would like to make this 5V compatible probe use the same firmware as the Rusty Probe. From what I have read in the firmware code, modifications will be needed to add 5V support but I'm sure it can be done in a way to maintain compatibility with existing Rusty Probes. From a hardware perspective the existing Rusty Probes can safely receive 5V signals but are not able to send them. I will follow up on firmware compatibility on a different issue on the firmware repository.

Finally I would like to express my thanks for your work and effort but into this project. I hope we can collaborate to further improve this project.

korken89, 2024-05-04:
Hi, thanks for the interest!

The calculations and possible output voltages are in the GNU Octave scripts in the repo that can be run with `octave-cli math/find_resistors.m`.
In here you can also change the resistors OR ask the script to find resistors for a new desired output range. The script is quite capable. :)
The first half is for solving, and the second half of the script is for plugging in "selected resistors" - as the solver does not follow E96 for example.

When it comes to 5V support there are 3 things that need to change:

1. The LDOs as you have noticed.
2. Voltage dividers for measuring voltage with the MCU.
3. The voltage translator ICs that have a requirement of Output signal <= Input signal (RP2040 3.3V)

And that should be it! 

BR Emil

acedogblast, 2024-05-04:
Thanks for the information. The current voltage divider for `T_VCC_MEAS` uses two 47K resistors which would divide the voltage in half. Half of 5V is 2.5V which should be fine the RP2040's ADC since the max voltage for the ADC input is 3.3V. Why do you say it needs to be changed?

korken89, 2024-05-05:
That's correct, I mixed it with another design.

acedogblast, 2024-05-10:
Another concern that I have found is that ~RESET for the target is not handled through a voltage translator. It is only pass through the NUF8401 100ohm chip then to the RP2040. A 5V native target could send 5V into the RP2040. In the firmware it does not seem that we are using ~RESET as an input but only as a "open drain" output. 

Also to note an open-drain output is an output that either a floating/unconnected hence "open" or the output is driven to GND which "drains" the output pin. Often such an output has a pull-up resistor to ensure the output is either high or low but in this case of the Rusty Probe it is not needed.

korken89, 2024-11-10:
If there are no more questions here I'm going to close this issue. Feel free to reopen if there are more questions!

closed 2024-11-10 by korken89