probe-rs/rusty-probe #18: https://shop.probe.rs is a scam?

WizardUli, 2024-07-28:
I ordered 2 rusty-probes on 24.1.2024 from the shop in https://github.com/probe-rs/rusty-probe/blame/d55237719e234000097c2dac1f8991a65e5a82e6/README.md#L41 got an order confirmation from shopify and since then there has been no communication whatsoever.

I would have understood if there were an email after a couple of months saying that there were problems with manufacturing or logistics or that the current hardware revision was too buggy and you were waiting for fixes and stabilization or that you had underestimated manufacturing of a real world product and could not fulfill the orders at all.
But this is either an unprofessional conduct or an outright scam.

Yatekii, 2024-07-28:
Many people have asked me directly and I tried to always offer insight as best as possible. Unfortunately I do this in my sparetime for your benefit and no financial gain and it's hard for me to always send out emails with ETAs that I cannot actually promise because as you can see we still did not ship and all the timings are flakey. If you asked me for your money back, you would have it tomorrow.

closed 2024-07-28 by Yatekii

WizardUli, 2024-07-28:
See, having this information at the shop instead of "lead time couple of weeks" would be enough. Any update at all even changing the wording to "a couple of months" would be ok.

If you ever get the production running well, consider raising the prices or just offering pricier variants where the difference goes into supporting you. Good luck.

Yatekii, 2024-07-28:
When I first put this up, it was a couple weeks. And estimate was always a couple weeks. Unfortunately I cannot guess all the unforeseen circumstances in advance.
I had plenty of people claim that the shop page did not state anything about delivery delays - which it does - making it obvious that putting more info there would not really help, especially since all the info I can put at any point in time is: The probes are being shipped in a couple weeks. After all the production shenanigans (samples being bad, Chinese new year, second batch of samples being lost in Swedish post for one month, etc) the probes were estimated to arrive at my place end of May, but then they were sent to Sweden and made a detour and I got them 2 weeks ago. They will be shipped this week.

trembel, 2025-02-17:
Would it be possible to put on kitspace, s.t. one-click orders could be out-sourced?