probe-rs/rusty-probe #20: rusty-probe enumerates two USB devices

igiona, 2024-09-09:
At least on windows (I did not test other OSs yet), the rusty-probe enumerates as two devices with the same serial number.
For this reason, `probe-rs` requires additional parameters `--probe 1209:4853` even though I actually have only one probe connected.
When developing and working via command-line it's quite annoying.

Is there a way to avoid this double-enumeration?
Why it's there in the first place? I don't see for me a way to select "the other" probe, since PID and VID are the same...

Yatekii, 2024-09-09:
cc @korken89

igiona, 2024-09-09:
Additional information, this the output of `probe-rs list` :

```shell
$ probe-rs list
The following debug probes were found:
[0]: Rusty Probe with CMSIS-DAP v1/v2 Support (VID: 1209, PID: 4853, Serial: DC645020138A1122EF4014, CmsisDap)
[1]: HS-Probe CMSIS-DAP v1 Interface (VID: 1209, PID: 4853, Serial: dc645020138a1122ef4014, CmsisDap)
```

jannic, 2024-10-01:
A similar thing has been mentioned on matrix a while ago: https://matrix.to/#/!vhKMWjizPZBgKeknOo:matrix.org/$9MlglhlFheOzD1NGEnnf1jRZrPQokowR8M-SNUTeMJY?via=matrix.org&via=chat.berline.rs&via=tchncs.de

Dirbaio diagnosed it as probably caused by https://github.com/kevinmehall/nusb/issues/22

korken89, 2024-10-04:
That's odd, how do you get this to happen? I was not able to reproduce on any of my computers, but it seems like two interfaces have different descriptions.
I'll look into it during my update of the HAL.

igiona, 2024-10-06:
> That's odd, how do you get this to happen? I was not able to reproduce on any of my computers, but it seems like two interfaces have different descriptions.
> I'll look into it during my update of the HAL.

Nothing special to be honest.
I'm using a USB HUB, hard to believe but maybe it's because of that?
My PC has WSL enabled, maybe the USB drivers get messed up?
I'll try on other machines as well and let you know.

korken89, 2024-10-06:
Ah sorry, I did not know you ran on Windows.
I'll dig up a Windows computer to test on :)

abelREK, 2024-11-05:
I can confirm that this also happens on windows.

korken89, 2024-11-10:
I've checked now and we only register one device, however it has multiple interfaces. Not sure why that would be a problem. To me it seems like an `nusb` issue on Windows, has anyone opened an issue at `probe-rs`? The interface logic seems like it needs special handling in Windows or maybe `nusb` needs some looking into. 😅

igiona, 2024-11-10:
> I've checked now and we only register one device, however it has multiple interfaces. Not sure why that would be a problem. To me it seems like an `nusb` issue on Windows, has anyone opened an issue [...]

Thank you for looking into it.
If it is a would be an USB driver issue, other probes would suffer from the same issue no?
I have a AK-CMSIS-DAP here, and it enumerates only one...

```bash
$ probe-rs.exe list
The following debug probes were found:
[0]: Rusty Probe with CMSIS-DAP v1/v2 Support (VID: 1209, PID: 4853, Serial: DC645020138A1122EF4014, CmsisDap)
[1]: AK-CMSIS-DAP (VID: 10c4, PID: 8b54, Serial: AKA5130B, CmsisDap)
[2]: HS-Probe CMSIS-DAP v1 Interface (VID: 1209, PID: 4853, Serial: dc645020138a1122ef4014, CmsisDap)
```

korken89, 2024-11-10:
Ah no, I see the confusion, but the difference is that the probe is one device with two interfaces. It exposes a CMSIS-DAP v1 and v2 at the same time so both old and new tooling can use it. The issue though is that probe-rs, AFAIK, hides redundant interfaces of probes and only shows the v2. And in here lies the issue because on Windows, for some reason, the interfaces don't get the same serial ID. Hence why it can be fixed in probe-rs by doing comparison that is not case sensitive, or in nusb to fix the serial ID output.

In your case though I would just remove the V1 interface, it would be the easiest solution. Maybe it's worth adding a feature gate to enable the dual interface that's enabled by default? It would make it easier.

igiona, 2024-11-10:
> Ah no, I see the confusion, but the difference is that the probe is one device with two interfaces. It exposes a CMSIS-DAP v1 and v2 at the same time so both old and new tooling can use it. The [...]

They are the same serial, but a different case.
Likely probe-rs does a full string match on the serial number, relaxing the check could solve the issue I guess.
I'll open an issue on probe-rs soon.

> In your case though I would just remove the V1 interface, it would be the easiest solution. Maybe it's worth adding a feature gate to enable the dual interface that's enabled by default? It would make it easier

With this you mean "customizing" the FW of the rusty probe, right?

korken89, 2024-11-10:
> With this you mean "customizing" the FW of the rusty probe, right?

Correct. It's a small change to perform.

closed 2024-11-22 by Yatekii